SolderedElectronics/Inkplate-micropython #4: setFont how to change font used with printText

michael3890, 2021-04-06:
Has anyone managed to change the font used in inkplate - micropython when writing to the inkplate display
have tried to use the setFont function .- but no success

nitko12, 2022-02-10:
Yes, setFont is part of the Adafruits CircuitPython library, primarily designed to be used with https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font which then requires their displayio class and that represents a problem, as Inkplate is not supported by them.

You could use https://github.com/peterhinch/micropython-font-to-py with some code to draw the generated bitmaps (essentialy modify the library) to achieve different fonts, also you could generate a dict with letters from .bcf files and use that probably, but theres are rather a workaround than a solutoin.

The main problem is that we're relying on machine module for most stuff as it's more low level, which circuitpython doesn't use in favor of Adafruit unified modules. 

We might switch to Adafruits modules soon tho, as there are more and more problems like this popping up :)

closed 2022-02-10 by nitko12

zachstence, 2023-01-28:
@nitko12 Any update on potential solutions here? I've spend today playing around with `micropython-font-to-py` and converting to the format used by Adafruits `font_dict`, but haven't had much luck.

It seems to me that the Adafruit `font_dict` should be an array of bytes, where each byte (excluding the first two) describes the pixels that are on/off for that column of the rendered character. However, since a byte must be <=255, that means each character can only be 8 pixels tall. This really limits the resolution of a font.

Am I missing something? Is there another way to use a custom font?

zachstence, 2023-01-28:
Aha, never mind! I've made some progress.

I converted a font with `micropython-font-to-py`, and adapted their `font_test.py` to get me the pixels to draw for the font. Then simply telling Inkplate to draw those pixels gets the font rendered properly.

Font to pixels: https://github.com/zachstence/inkplate-todo/blob/main/inkplate-todo/src/fonts/fonts.py
Drawing pixels from font: https://github.com/zachstence/inkplate-todo/blob/main/inkplate-todo/src/main.py