SolderedElectronics/Inkplate-micropython #13: RFC: Inkplate device type detection

kratz00, 2021-11-16:
It seems for every device the _Inkplate_ class (~ 1000 LOC) and all the examples are duplicated, with only a few minor exceptions.
I could not find any information, but is there any way to detect the device type programmatically (preferable using non device specific code)?

If not a software solution might work (disclaimer: Quickly made up proposal):

- a base class _InkplateBase_
- one _Inkplate_ class derived from _InkplateBase_ per device type in its own Python module (e.g. inkplate10.py)
- copy the base class and the class for your device type to the board
- use the following code to load the right implementation of the _Inkplate_ for your device

```
def load_inkplate_module():
    global Inkplate
    for module_name in ['inkplate6', 'inkplate10']:
        try:
            module = __import__(module_name, globals(), locals())
            Inkplate = module.Inkplate
            return
        except ImportError:
            pass
    raise ImportError('No Inkplate module found')

load_inkplate_module()

inkplate = Inkplate(Inkplate.INKPLATE_1BIT)
```

Unfortunately *importlib* is not available in standard MicroPython, so this code (or something better) could be added in some module,
so it can be imported and re-used.

This approach could help to reduce a lot of the code duplication and to unify all/most of the examples.

kratz00, 2021-11-23:
Or maybe even a less complicated approach, which does not require magic code to load the _Inkplate_ module.
The steps are nearly the same:

- a base class InkplateBase
- one Inkplate class derived from InkplateBase per device type in its own Python module (e.g. inkplate10.py)
- copy the base class
-  copy the class for your device type to the board, **but** rename to generic name *inkplate.py*

For example:

`./pyboard.py --device /dev/ttyUSB0 -f cp inkplate10.py :inkplate.py`

davidzovko, 2021-11-24:
Sorry that we are slow. @nitko12 will check on this soon enough. Thank you for the inputs!

nitko12, 2022-02-10:
`./pyboard.py --device /dev/ttyUSB0 -f cp inkplate10.py :inkplate.py`

Is a nice solution, but it would not solve all the examples beacuse they have hard coded sizes for elements for the sake of simplicity (we'll might be adding more complicated examples where that would be a problem)

As the Inkplate is designed for beginners, adding more complicated code would make it hard for them to understand and subsequently modify code if they wish.

A solution might be, if you wish to, to contribute a more advanced Inkplate Python module, it can be on your Github, which will be linked in our "beginner" friendly repo :)

Hope this explains the design choices in the module, thanks for the ideas and contribution.