SolderedElectronics/Inkplate-micropython #17: Battery voltage from display.readBattery() is zero on an Inkplate 10

jeffam, 2022-05-04:
When I run the [most recent battery and temperature example](https://github.com/e-radionicacom/Inkplate-micropython/blob/6c18b42af23a8da2dae815daee076670f6150f27/Examples/Inkplate10/batteryAndTemperatureRead.py) on my Inkplate 10, I get a value of `0.0` for the battery voltage.

I run this via `python3 lib/pyboard.py --device /dev/cu.usbserial-1420 lib/Examples/Inkplate10/batteryAndTemperatureRead.py`. `lib/` is a directory with the `.py` files from this repo.

The [Arduino battery and temperature test](https://github.com/e-radionicacom/Inkplate-Arduino-library/blob/93394781b203303727c72e9e3d872dcb74a8a091/examples/Inkplate10/Advanced_Inkplate_Features/Inkplate_Battery_Voltage_And_Temperature/Inkplate_Battery_Voltage_And_Temperature.ino) for the Inkplate 10 _does_ return a value: `4.37` with a newly installed lipo battery.

Any advice for troubleshooting? Am I missing some configuration for MicroPython?

kratz00, 2022-12-20:
@jeffam Try switching the pin logic

1.  [Set to high](https://github.com/SolderedElectronics/Inkplate-micropython/blob/master/inkplate10.py#L145)
2. [Set to low](https://github.com/SolderedElectronics/Inkplate-micropython/blob/master/inkplate10.py#L149)

Should look like this then:

```
    # Read the battery voltage. Note that the result depends on the ADC calibration, and be a bit off.
    @classmethod
    def read_battery(cls):
        cls.VBAT_EN.value(1)
        # Probably don't need to delay since Micropython is slow, but we do it anyway
        time.sleep_ms(1)
        value = cls.VBAT.read()
        cls.VBAT_EN.value(0)
        result = (value / 4095.0) * 1.1 * 3.548133892 * 2
        return result
```

daemonhorn, 2023-07-23:
Original default `soldered_inkplate10.py` module for read_battery (does not report any voltage: `0.0`):
```
# Read the battery voltage. Note that the result depends on the ADC calibration, and be a bit off.
    @classmethod
    def read_battery(cls):
        cls.VBAT_EN.digitalWrite(0)
        # Probably don't need to delay since Micropython is slow, but we do it anyway
        time.sleep_ms(1)
        value = cls.VBAT.read()
        cls.VBAT_EN.digitalWrite(1)
        result = (value / 4095.0) * 1.1 * 3.548133892 * 2
        return result
``` 
Modified pin logic 0->1 and 1->0  using .digitalWritet() as below in `soldered_inkplate10.py` (works and reports reasonable voltage: `4.4` on charger and `4.2` with full charge on 3000mAh 3.7v battery):
I may manually tweak the calculations a bit since wth a good multimeter, I am reading `4.04` when the result shows `4.2`
```
# Read the battery voltage. Note that the result depends on the ADC calibration, and be a bit off.
    @classmethod
    def read_battery(cls):
        cls.VBAT_EN.digitalWrite(1)
        # Probably don't need to delay since Micropython is slow, but we do it anyway
        time.sleep_ms(1)
        value = cls.VBAT.read()
        cls.VBAT_EN.digitalWrite(0)
        result = (value / 4095.0) * 1.1 * 3.548133892 * 2
        return result
```
Can someone independently verify that this patch works on "more than just my board" , and please commit to `soldered_inkplate10.py` on dev branch for inclusion in the next release  (combined with above patch for original `inkplate10.py`?

svanya, 2023-08-30:
Same problem with soldered_inkplate6_PLUS.py, changing the pin logic (0->1, 1->0) helped.

svanya, 2023-09-04:
> Same problem with soldered_inkplate6_PLUS.py, changing the pin logic (0->1, 1->0) helped.

hmm, but when I disconnect the battery, it still shows the value: 4.471949V

rsoric, 2023-09-05:
Hi @svanya , this was fixed on the dev branch recently and will soon be pushed to main. If you disconnect the battery, the 4.47V reading is also expected because of the way the battery charging chip works (it's the same on Arduino).

closed 2023-09-05 by rsoric