SolderedElectronics/Inkplate-micropython #18: Inkplate File is too large

markus-seidl, 2022-07-18:
I've tried to upload the inkplate6plus file to the device but got errors while uploading. The smaller files (like the mcp one) uploaded flawlessly. 
So my guess is it's too large. Additionally, from a clean code standpoint, a file with over a 1000 line is hard to understand.

kratz00, 2022-12-20:
@markus-seidl Please add more details:

- what MicroPython version are you running
- what command did you use to upload files
- what is the error message you get

markus-seidl, 2022-12-20:
Sorry, but I abandoned the project completely and moved to a different product.

closed 2022-12-20 by markus-seidl

tritip, 2022-12-20:
What product please?— c

markus-seidl, 2022-12-20:
I was totally unsatisfied with this python implementation, so I've moved one up to a kindle tablet (which is equally expensive) and used this as a simple wall mounted display. It has a lot of different caveats for my use case, but it was tremendously simpler to program.