SolderedElectronics/Inkplate-micropython #19: Problems with Inkplate Color examples

hermannmarkus, 2022-09-30:
Hi there,

I recieved my Inkplate Color today and want to try out the micropython capabilities. I followed the steps erasing the flash, installing the current firmware from this repo and copied the library files onto the Inkplate. But none of the examples are working for me.

```
python3 pyboard.py --device /dev/tty.usbserial-110 "Examples/Inkplate6COLOR/basic.py"                     
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
  File "inkplate6_COLOR.py", line 146, in begin
  File "mcp23017.py", line 18, in __init__
  File "mcp23017.py", line 30, in write
OSError: [Errno 19] ENODEV
```

Is there anything I may have done wrong?

When I should provide additional information please let me know.

zond, 2022-10-07:
I have the exact same problem.

tithonium, 2022-11-01:
Same issue.

kratz00, 2022-12-20:
The code in question is initializing the MCP23017, which I guess does not exist on the Inkplate 6COLOR, judging by this [picture](https://e-radionica.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/_/5/_562534-edit.jpg) it uses a different I/O expander IC, namely a [NXP PCAL6416AHF](https://www.nxp.com/docs/en/data-sheet/PCAL6416A.pdf).

This can be easily confirmed by running the following code.

```
from machine import I2C, Pin
i2c = I2C(0, scl=Pin(22), sda=Pin(21))
print(i2c.scan())
```

What output do you get?

zachstence, 2023-01-04:
> The code in question is initializing the MCP23017, which I guess does not exist on the Inkplate 6COLOR, judging by this [...]

This is the output I get.
```py
[32, 81]
```

zachstence, 2023-01-04:
I had reached out to their customer support when I originally encountered this problem and they said they were looking into it. Here's the email thread for anybody interested.
[image]

I don't mean to pester, but Inkplate members (@davidzovko, @nitko12, @brunomatkovic, @BornaBiro, @ifadiga) do you have any updates on this issue? Any ways that the community can help out?

kratz00, 2023-01-04:
Judging from the output your hardware has a MCP23017. 
Maybe there are different hardware revisions of the Inkplate 6COLOR (in this picture https://inkplate.readthedocs.io/en/latest/_images/inkplate6color_back.jpg it has a MCP23017 vs. https://e-radionica.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/_/5/_562534-edit.jpg - here it has a PCAL6416AHF)

@zachstence what does it say for your board - look at the upper right corner?

ifadiga, 2023-01-04:
Hello,

Sorry for the late response.

We designed Inkplate 6COLOR with the MCP23017 GPIO expander but because of IC shortage, we must change the GPIO expander to PCAL6416A. 

We working on the micropython code for the new GPIO expander and soon it will be available for Inkplate 6COLOR.

kratz00, 2023-01-04:
@ifadiga thanks for your comment.

One thing I do not understand - shouldn't the PCAL6416A have a different I2C slave address? So why does a scan find '32' aka 0x20, the slave address of a MCP23017 in @zachstence case?

BornaBiro, 2023-01-04:
@kratz00 Actually PCAL6416A and MCP23017 have exactly the same base I2C address (0x20) as stated in the [datasheet](https://www.nxp.com/docs/en/data-sheet/PCAL6416A.pdf), page 7. But the internal registers of these two I/O expanders are completely different.

kratz00, 2023-01-04:
@BornaBiro Thanks again for the clarification.

zachstence, 2023-01-04:
I do have the purple board with the PCAL6416A

kratz00, 2023-01-22:
Support for the PCAL6416A was added in the *dev* branch - https://github.com/SolderedElectronics/Inkplate-micropython/tree/dev
As there was no official communication yet, I am not sure about the state of the implementation,
but I guess its worth a shot.
Try if you want do and give some feedback I would say :smile:

ifadiga, 2023-01-22:
Hello, 

We work on micropython library for Inkplate. We didn't publish it on master because we need to test it.

You can use it if you wish, but we can't be sure that everything works until we test it.

zachstence, 2023-01-24:
The PCAL6416A adaptations made on the `dev` branch work for me! I was able to successfully run `Examples/Inkplate6COLOR/basic.py` on my Inkplate!

@ifadiga let me know if I can help out with any testing using my hardware.

closed 2023-09-05 by rsoric