SolderedElectronics/Inkplate-micropython #23: Insufficient memory to initialize screen buffer

ebrynne, 2023-05-11:
I have an Inkplate6PLUS, running micropython 1.20, and all examples fail upon calling `display.init()` because the `self._framebuf = bytearray(D_ROWS * D_COLS // 4)` allocation asks for more memory than the device has available. 

As an example:
```
❯ python3 pyboard.py --device /dev/tty.usbserial-110 "Examples/Inkplate6PLUS/frontlight.py"

Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "inkplate6_PLUS.py", line 950, in begin
  File "inkplate6_PLUS.py", line 647, in __init__
MemoryError: memory allocation failed, allocating 194048 bytes
```

Stepping through manually in a repl, it's clear the micropython environment is heavily memory constrained and the failure seems reasonable:

```
>>> import 
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 42528, free: 68640
 No. of 1-blocks: 481, 2-blocks: 61, max blk sz: 451, max free sz: 440
>>> display = Inkplate(Inkplate.INKPLATE_1BIT)
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 47488, free: 63680
 No. of 1-blocks: 499, 2-blocks: 62, max blk sz: 451, max free sz: 440
>>> display.begin()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "inkplate6_PLUS.py", line 950, in begin
  File "inkplate6_PLUS.py", line 647, in __init__
MemoryError: memory allocation failed, allocating 194048 bytes
```

ebrynne, 2023-05-11:
Ah crap, it's because the tutorial links to `https://micropython.org/download/esp32/` instead of `https://micropython.org/download/esp32spiram/` for getting the builds from the source (the included version is 1.18, two releases behind). I'll find time to submit a PR to fix the docs.

rsoric, 2023-05-15:
No need, just updated them based on your suggestion. Thank you for letting us know!

closed 2023-05-15 by rsoric