SolderedElectronics/Inkplate-micropython #26: ImportError: no module named 'PCAL6416A' - Soldered Inkplate 10

SMores, 2023-06-03:
Hello! I have a new Soldered Inkplate 10. I followed the instructions in the README to install micropython on the inkplate:

- `python3 esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 esp32spiram-20220117-v1.18.bin`
- `python3 pyboard.py --device /dev/ttyUSB0 -f cp mcp23017.py soldered_inkplate10.py image.py shapes.py gfx.py gfx_standard_font_01.py :`

When I try to run the example, however, I get the following error:

```console
$ python3 pyboard.py --device /dev/ttyUSB0 "Examples/Soldered_Inkplate10/basicBW.py" 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "soldered_inkplate10.py", line 8, in <module>
ImportError: no module named 'PCAL6416A'
```

Any suggestions?

SMores, 2023-06-03:
Aaaand upon closer inspection, of course I just needed to also copy the `PCAL6416A.py` file from this repo onto the device. All set now!

closed 2023-06-03 by SMores

rsoric, 2023-06-06:
Hi @SMores 

Great to hear you got it working, we have recently updated the README to make this more clear.

-Rob