SolderedElectronics/Inkplate-micropython #28: ENODEV when running basicBW example on 6plus

halffullheart, 2023-11-02:
I have followed the [docs steps 1 - 3](https://inkplate.readthedocs.io/en/latest/get-started.html#micropython) and now am trying to run the `basicBW.py` example like so:

```
python3 pyboard.py --device /dev/tty.usbserial-110 examples/basicBW.py
```

I get the following error:
```
Traceback (most recent call last):
  File "<stdin>", line 17, in <module>
  File "soldered_inkplate6PLUS.py", line 975, in begin
  File "soldered_inkplate6PLUS.py", line 76, in init
  File "PCAL6416A.py", line 84, in __init__
  File "PCAL6416A.py", line 110, in read
OSError: [Errno 19] ENODEV
```

That sounds like the PCAL6416A isn’t found.

Indeed my board came with a note reading:
> Due to the current chip shortage in the world, the MCP2307 below had to be removed so you could get this Inkplate.
> The GPIO pins below aren’t usable. To make up for it, here’s a code for 10% off your next order.

That note does not mention a PCAL6416A and indeed looking at the board, there appears to be no chip at all in the space labeled for the MCP2307. I assume this means my board has neither a MCP2307 or PCAL6416A.

If that’s true, can I get things working with micropython with different code than what is in soldered_inkplate6PLUS.py?

rsoric, 2023-11-06:
Hi @halffullheart 

sorry for a bit of a late reply, was out of office. Your board will either have one expander or the other - but if you have a note which mentions the MCP expander - I assume that one goes for your board. Your board will have one MCP though, and that one is used for driving the ePaper. The note is referring to the secondary GPIO expander usually found on Inkplate boards - allowing you to use additional GPIO. 

So, do, when you're copying your libraries after flashing the firmware, make sure not to copy the 'Soldered' version of the board as those use the PCAL expander, but copy the other one, and also of course copy mcp23017.py instead of PCAL6416A.py. 

Give this a shot and see if it works. Sorry if it's a little bit confusing, I tried to be as clear as possible in the README.

-Rob

halffullheart, 2023-11-17:
I haven’t had a chance to test with micropython yet but I did solve some issues with Arduino IDE by using the e-radionica profile instead of the “soldiered” one, so I think what you’re suggesting will also work with micropython. Thank you!

rsoric, 2023-11-20:
Glad to hear! When you do, please write back here so I can close the issue.

closed 2026-01-15 by JosipKuci