SolderedElectronics/Inkplate-micropython #30: sdcard.py missing

jwefers, 2023-12-26:
`https://inkplate.readthedocs.io/en/latest/get-started.html#micropython` sows copying lib files to the device. However, `sdcard.py` missing in the repo.

rsoric, 2024-01-02:
Hi @jwefers ,

would you please point me to where this file would be required? The SD Card code is in the 'machine' module in the .bin of the micropython firmware. You don't need to copy sdcard.py to use the sdcard. 

Please give one of the examples a try, if I'm missing something please let me know and I'll happily write a follow-up and test any example which is required.

-Rob

jwefers, 2024-01-02:
step 3:

`python3 pyboard.py --device /dev/ttyUSB0 -f cp mcp23017.py sdcard.py inkplate6.py image.py gfx.py gfx_standard_font_01.py`

rsoric, 2024-01-02:
Ah, I see, the documentation is different on readthedocs and in this repository. That's my mistake and I apologize, going to correct it right now. Please use the guide provided in this repository.

My best guess is that this file was required in an older version.

closed 2024-01-02 by jwefers