SolderedElectronics/Inkplate-micropython #32: Font format?

mbanzi, 2024-01-14:
I'm using Micropython on the Inkplate 6 and I realised the font format doesn't seem to follow any of the formats normally used to represent fonts in Micropython or Arduino code.

Is there any documentation on how to convert Fonts to your format??

rsoric, 2024-01-16:
Hi @mbanzi 

We weren't playing around with custom fonts in Python but I've had a look at gfx_standard_font_01.py, It shows a comment with letter format and Key format. The first two bytes are the height and width of each character, followed by the bitmap, where "1" represents a drawn pixel. Let's have a look at the letter A. It's ?DDD?
Ascii to binary gives us:
? -> 0b00111111
D -> 0b01000100

And now, if we paste that in VScode and CTRL+F all the '1's for easy visualisation, we get this: 
[image]

So that's how it works.

rsoric, 2024-01-31:
Closing as there doesn't seem to be more activity on this.

@mbanzi if you have any more questions, don't hesitate to let us know via here or our [support](https://soldered.com/contact/).

closed 2024-01-31 by rsoric