SolderedElectronics/Inkplate-micropython #33: README instructions for Windows are incorrect.

b-morgan, 2024-01-15:
The Windows examples use `python esptool.py` but esptool.py doesn't exist in the directory. esptool.exe does exist so substituting `esptool` for `python esptool.py` and the rest of the command line works.

rsoric, 2024-01-16:
Hi Brad,

esptool.py should exist in the directory, as described in step 3, and I've personally ran these commands to get microPython to work. I'll add your comment as a side-note in case anyone else has problems. Thanks a lot or pointing it out!

The whole installation process of microPython is a bit complicated, we know, and we do have a solution for this in mind (some sort of app with a GUI which helps users run these commands), just have to get around to it.

-Rob

closed 2024-01-16 by rsoric

b-morgan, 2024-01-16:
Perhaps I wasn't clear in my initial post, Step 1 says:
`Install esptool - the command line tool used to upload firmware to the ESP32. Get it from [here](https://github.com/espressif/esptool)`.

On the main page, you can go to [Releases](https://github.com/espressif/esptool/releases) and for Windows you get `esptool-v4.7.0-win64.zip`. This .zip file does NOT contain esptool.py (or any other .py files), it contains esptool.exe (and other .exe files).

Since step 1 doesn't explicitly say download this repository I went with the Releases option. If you clone or copy the GitHub repository, then you do get all the source files including `esptool.py`.

I'd guess that by distributing .exe files, it removes the need to install Python on the Windows host (but Python will be needed later). I could find no documentation on the GitHub site that mentions .exe files but after doing a Google search for "esptool.exe" I did find some references.

Now that you know how I (and perhaps others) obtained `esptool` you could fix step 1 and your clarification on the Windows examples. After thinking about it I decided to take a stab at modifying the README.md. Note: I split Step 1 into two steps and I modified the final example removing `// This one might need to be started twice` and replacing it with `python pyboard.py --device COM5 -f ls`. This command is the sacrificial one that fails but if it doesn't fail, it just lists the files currently present.

[README.zip](https://github.com/SolderedElectronics/Inkplate-micropython/files/13952063/README.zip)