SolderedElectronics/Inkplate-micropython #35: Can't read battery level of Inkplate6COLOR

besi, 2024-01-22:
I'm running the following code in order to read the battery level:

```python
from inkplate6COLOR import Inkplate
Inkplate.readBattery()
```

I will get the following error:

```
Inkplate.readBattery()
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "/lib/inkplate6COLOR.py", line 473, in readBattery
AttributeError: type object 'Inkplate' has no attribute 'VBAT_EN'
```

besi, 2024-01-23:
Apparently I have the version 2.2.0 from around Januar 2023

[image]

rsoric, 2024-05-28:
Hi @besi 

really sorry for such a late reply, as we don't prioritize this repository. I've just made a push to dev where this issue is fixed on inkplate6COLOR, check out the example as well: https://github.com/SolderedElectronics/Inkplate-micropython/blob/dev/Examples/Inkplate6COLOR/batteryRead.py

Hopefully that works.

closed 2024-05-28 by rsoric