SolderedElectronics/Inkplate-micropython #36: readBattery always returns 0.0 on inkplate5

Danielside, 2024-02-13:
So I continue tinkering with the device. With a battery that has been charging for about 2 hours, I try to run this simple example

```
from inkplate5 import Inkplate
from soldered_logo import *
import time

display = Inkplate(Inkplate.INKPLATE_1BIT)

if __name__ == "__main__":
    display.begin()
    display.clearDisplay()
    print(display.readBattery())
```

Being the result always "0.0".

I tried to debug a bit by making the function read_battery return only the value, here:

```
    @classmethod
    def read_battery(cls):
        cls.VBAT_EN.digitalWrite(0)
        # Probably don't need to delay since Micropython is slow, but we do it anyway
        time.sleep_ms(5)
        value = cls.VBAT.read()
        print(value)
        cls.VBAT_EN.digitalWrite(1)
        #result = (value / 4095.0) * 1.1 * 3.548133892 * 2
        result = value
        return result
```

And `result` always has the value 0, I understand that the reading has been incorrect in the first place. I have tried to give it a little more time (sleep_ms(15)) to no avail.

Danielside, 2024-05-22:
So, it seems that you care more about selling new devices than to maintain those that you already sold. Besides, I don't know if it is normal that a simple watch that to partial resets and sleeps most of the time https://git.sr.ht/~danielside/Reloj/tree/master/item/main.py drains the battery in an hour.

These devices and their API are a complete dissapointment.

rsoric, 2024-05-28:
Hi @Danielside 

we prioritize the Arduino library over this one. The Micropython library was originally written as a contribution so, we're just adapting to it. If you're looking to use Inkplate in a more serious project with requirements, I recommend the Arduino library over this.

Could you try changing the first VBAT value to 1 and the second to 0? I believe that should be the way the GPIO pin should interact with the MOSFET.

rsoric, 2024-05-29:
Hi @Danielside

really sorry about this. I understand how this was frustrating for you. I take responsibility for it as I'm the one maintaining this repository, let me know if there is some other feature of this I could assist you with and I'd be happy to. 

Unfortunately, we have a limited number of hours spent working on our libraries so we have to prioritize the Arduino one, this is mainly why this happened.

I've gotten around to look at this and the issue was an inverted usage of the VBAT_EN pin, check out the latest push to the dev branch: https://github.com/SolderedElectronics/Inkplate-micropython/tree/dev - if you're still interested in using the library, give it a try.

-Rob

Danielside, 2024-06-08:
It works now, thanks.

When I bought the device, I would have appreciated a BIG NOTE besides the micro python library, saying that it is not thoroughly tested or officially supported.

Keep in mind that you have been in the market for several years now, I did not expect this library to have these kind of problems. 

Daniel

closed 2024-10-22 by rsoric