SolderedElectronics/Inkplate-micropython #38: How to schedule functions at defined times like Cron-Jobs

mfasold, 2024-10-04:
I would like to run some functions at defined times and dates. Example:

- on Monday to Friday at 7:00 and 8:00 run function A
- on Saturday and Sunday at 10 run function B

I understand that the RTC allows me to get the time. How can I use this to run the functions at the defined times and sleep in between? Can this be done using available libraries (preferred), or can you recommend external libraries?

rsoric, 2024-10-07:
Hi @mfasold ,

I'll have to check regarding your other opened issue regarding the RTC on Inkplate 6COLOR. Could you please take a picture of the back side of your board so I can verify the version which you have? This will help me resolve the issue faster.

As a side note, we mainly maintain and update the Arduino library, this one was done originally as a user contribution and does have the key features but I would personally recommend using the Arduino one.

-Rob

mfasold, 2024-10-07:
Thank you for the reply. I was having the same impression regarding Arduino vs Micropython. Maybe I'll switch. But if the RTC works, this might be enough for me. I found the `micropython-mcron` library, which might be the solution to the CRON question.

Please find the image here.
[image]

rsoric, 2024-10-14:
Hi @mfasold ,

I've corrected some things in the code and was able to load from the RTC:
[image]

Long story short, it's due to the fact there is a different implementation of the display with this Inkplate compared to the others so the Python class structure is different, and as we've had the RTC implemented in others, this one required some corrections in the implementation. Thanks for pointing it out to us. Try the [dev branch](https://github.com/SolderedElectronics/Inkplate-micropython/tree/dev) now please and if it's correct on your device as well, I'll push it to the main branch.

-Rob

rsoric, 2024-10-15:
Great to hear, thanks for letting us know about this, It will be pushed to main soon. So, from here, you can use the RTC's data to get the time and then periodically check it, and run a task if a time condition has been met. Gonna close this issue now.

closed 2024-10-15 by rsoric