SolderedElectronics/Inkplate-micropython #39: RTC example for Inkplate 6 Color does not work

mfasold, 2024-10-07:
This repository contains [example code for setting the RTC time](https://github.com/SolderedElectronics/Inkplate-micropython/blob/master/Examples/Inkplate6COLOR/RTC.py).

Firstly, there is an error on line 8:
```
>>> display = Inkplate(Inkplate.INKPLATE_1BIT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: type object 'Inkplate' has no attribute 'INKPLATE_1BIT'
```

I assume that `display = Inkplate()` will be a sufficient replacement.

When I try to set the time as in the example, there is an error

```
>>> display.rtcSetTime(9,39,10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "inkplate6COLOR.py", line 589, in rtcSetTime
NameError: name '_Inkplate' isn't defined
```

This class seems to be defined in the modules for other Inkplate devices? Could  it be added for the Inkplate 6 Color?

rsoric, 2024-10-07:
Hi @mfasold ,

sorry about the error in line 1, you're correct to assume your updated code is right as Inkplate 6COLOR has only one mode of operation.

As for rtcSetTime, I will have a look as soon as possible and see if it's possible to replicate and solve the issue.

rsoric, 2024-10-15:
Hi @mfasold ,

I've just ran this example and got it working:
```
C:\repos\Inkplate-micropython>python pyboard.py --device COM10 "Examples/Inkplate6COLOR/RTC.py"
{'year': 2024, 'minute': 39, 'hour': 9, 'day': 9, 'weekday': 5, 'month': 2, 'second': 10}
{'year': 2024, 'minute': 39, 'hour': 9, 'day': 9, 'weekday': 5, 'month': 2, 'second': 20}

C:\repos\Inkplate-micropython>
```

It was working, this is using the latest dev branch. Here's how I copied all the required files:
```
C:\repos\Inkplate-micropython>python pyboard.py --device COM10 -f cp inkplate6COLOR.py gfx.py gfx_standard_font_01.py mcp23017.py PCAL6416A.py image.py shapes.py soldered_logo.py :
```

So I'm not sure, but running this example has produced some display issues, which could be related to other things in the library.

Our main library is the [Arduino one](https://github.com/SolderedElectronics/Inkplate-Arduino-library), so I would suggest trying to use that for coding for the time being. The micropython library was originally a user contribution and we adapted it to different boards, but the Arduino library contains more features and is more updated, so that's what I'd reccoment.

rsoric, 2024-10-22:
Closing due to inactivity, please feel free to write back if required

closed 2024-10-22 by rsoric