SolderedElectronics/Inkplate-micropython #42: Error with function setFrontlight with Inkplate 6PLUS

Rorsc4achhh, 2025-03-04:
Tried with function Frontlight display.frontlight(1), which works great to turn on and off. 
However when using display.setFrontlight(30) got error:

pi@raspberrypi:~ $ python3 /home/pi/Desktop/Soldered_6PLUS/Inkplate-micropython-master/pyboard.py --device /dev/ttyUSB0 "/home/pi/Desktop/Soldered_6PLUS/Inkplate-micropython-master/Examples/Inkplate6PLUS/basicBW.py"
Sd card could not be read
Mono: clean 1325ms (26ms ea), draw 522ms (87ms ea), total 1847ms
Mono: clean 1324ms (26ms ea), draw 521ms (86ms ea), total 1845ms
Traceback (most recent call last):
  File "<stdin>", line 42, in <module>
  File "inkplate6PLUS.py", line 1334, in setFrontlight
  File "inkplate6PLUS.py", line 184, in setFrontlight
OSError: [Errno 19] ENODEV

Using Inkplate6PLUS related library, without soldered_

Rorsc4achhh, 2025-03-06:
changed the setFrontlight function in 6PLUS to :

    @classmethod
    def setFrontlight(cls, value):
        value = (63 - (value & 0b00111111))
        cls._i2c.writeto(FRONTLIGHT_ADDRESS, bytes([0, value]))

closed 2025-03-06 by Rorsc4achhh

rsoric, 2025-03-13:
Hi @Rorsc4achhh ,

glad you managed to fix it for yourself. If we get around to updating this repository we will keep this fix in mind.